Duet3D/Duet3Expansion #10: [FeatureRequest]: Closed Loop - Feedforward

lirwin3007, 2021-09-17:
### Is your feature request related to a problem? Please describe.

When in closed loop mode, the most error is introduced when changing acceleration.

### Describe the solution you propose.

Given that we know when we're going to be changing acceleration, we could add in a feed-forward term to try to account for this, and hopefully reduce our error on acceleration change

### Describe alternatives you've considered

N/A

### Provide any additional context or information.

N/A

T3P3, 2023-04-23:
@dc42  added the following to 3.5b2:

[Duet 3 EXP1HCL] [Duet 3 M23CL] Added experimental velocity feedforward parameter V to the existing M569.1 PID parameters. The default value is zero. Note, the standard step tuning move does not use this parameter, therefore if you use a nonzero value then you need to use a regular G1 move instead when you perform PID tuning.

dc42, 2023-04-28:
I have now added an acceleration feedforward term too. The PID step tuning manoeuvre should be updated to use a finite velocity and acceleration instead of a sharp step. The acceleration and velocity parameters need to be added to the M569.6 command.

closed 2023-09-25 by dc42