Duet3D/Duet3Expansion #22: Closed Loop with Linear CoreXY Encoders

skuep, 2024-08-04:
Hi everyone,
I am in the process of building a Pick-And-Place machine (OpenPNP) and I am currently using the excellent Duet Mini 5+. I was thinking about the feasibility to go Closed Loop for the CoreXY kinematics that I chose, which generally does not seem to be an issue (when using rotary encoders!).

However I was also thinking about using chinese linear encoders (E.g. Sino KA200) for even better positional accuracy and I am evaluating whether such a setup is possible: Closed Loop on CoreXY with A/B motors in conjunction with X/Y linear encoders.

There was a quick discussion already in the forums (https://forum.duet3d.com/topic/36201/corexy-linear-encoders-1hcl-closed-loop-expansion-board/) and the consensus is, that it is currently not possible. After a few hours looking through the code, I would pretty much confirm that this is currently not possible, since the encoder is always directly mapped to the motor, way after any kinematics calculations have been carried out.

I am thinking about tackling this in the future (sounds like a nice challenge) and so far I have devised the following plan:
1. Get a EXP3HC board (a board with more than one motor driver) so the firmware has access to two local motor drivers. 
2. Make firmware support CLOSED_LOOP feature. One encoder will run on the PDEC hardware and another encoder will be implemented using CCL. I think this should be possible, there is an appnote on the old AT(x)mega processors on exactly this. I think it could be made to work on the SAME5X (https://ww1.microchip.com/downloads/en/Appnotes/Interf-Quad-Encoder-CCL-w-TCA-TCB-DS00002434C.pdf).
3. Validate functionality using two regular rotary encoders on motor A/B connected on one of the expansion headers. From a quick glance, the required pins (PDEC and CCL) seem to be on the SPI1 header. +5V/GND could be sourced from another header
4. Implement a "encoder forward matrix" (or reverse matrix) feature into the expansion firmware that defines the relationship between encoders and drives. Currently each drive has their own encoder attached. This would need to be reworked a bit, so that one driver can access all (both) encoder positions and mix&match them according to the forward matrix. Ideally the user should be able to configure this encoder forward matrix using the M569.1 command similar to the M669 command.
5. Change the closed loop code (and tuning code) to get encoder values through the forward (or reverse) matrix so that each PID motor controller sees only decoupled "virtual" axes that they can use for control

I think I could make 1-3 work on my own, since it's basically a bit of hardware fiddling and modifying the expansion firmware. However for points 4-5 I am too inexperienced with the RRF stuff to be able to implement this. Also I am not yet sure if there are some caveats when implementing the encoder decoupling into the ClosedLoop class. Are there situations where the coupling is unwanted (e.g. encoder tuning? encoder calibration?). Issue https://github.com/Duet3D/Duet3Expansion/issues/11) could be related to exactly this question.

What do you think? Is this of enough interest to get one or two persons to help or is this too niche?

skuep, 2024-10-13:
Just a heads up, after I set up the build environment I noticed that the TMC5160 does not have the XDIRECT register. Bummer.

Maybe I need to find another way to get to my goal, since interest here seems also quite limited. Thanks anyway for all the work that went into this!

dc42, 2024-10-14:
TMC5160 drivers do have the XDIRECT register although I think it may have a different name. Implementation of phase stepping (which also used XDIRECT) is currently being implemented on the 3HC in the 3.6-dev branch of RRF.

skuep, 2024-10-14:
Thanks for the reply. I quickly grepped through the datasheet yesterday and couldn't find anything related. In the code, it says something about 0x2D, which is not the DIRECT register but the XTARGET register. 

So I am not sure, I think the mechanism works quite differently to the TMC2160, so it's good to hear that it might be supported soon, because I couldn't have managed that myself.

dc42, 2024-10-15:
Hi Simon,

From page 33 of the TMC5160A datasheet rev 1.14:

[image: image.png]

Best regards - David

skuep, 2024-10-15:
Wow, that's well hidden 🤣 alright that clears things up. Might have another go at it in the next days.

skuep, 2024-10-19:
In the first step I am trying to get closed-loop working on the stepper controller 0 and so far it compiles and I get the firmware binary.
See https://github.com/Duet3D/Duet3Expansion/compare/3.6-dev...skuep:Duet3Expansion:3.6-dev-3hc-closedloop?expand=1

Two questions (if you don't mind):
1. I noticed that on the 3HC board, there is no CLK being provided to the TMC5160, whereas on the 1HCL, a clock is provided to the TMC2160 controller. What's the idea behind that? Does the clock need to run synchronous? Do you think it's problematic that the 3HC doesn't provide a clock to the stepper controller?
2. Can I run the newly compiled firmware (based on 3.6-dev) on the 3HC board and connect it to the Duet3 Mini5 running an older firmware (I think 3.5-dev but need to check) or does the mainboard also need to run 3.6-dev?

dc42, 2024-10-25:
1. The purpose of providing a clock from the MCU to the TMC2160A on the 1HCL is to allow us to use a higher SPI speed to the TMC2160A. The datasheet gives a maximum frequency when the internal clock is used, and a higher frequency when an externa clock is used and the SPI clock is synchronised to it.
2. You might get away with running 3.5 on the main board if you don't have any filament monitors connected to the 3HC and you don't mind having no input shaping on the 3HC.